dorukeker/gyronorm.js #9: Support on windows phone

CrowdHailer, 2015-05-01:
Hi, I seam to get no calls to the start call back on my windows phone. This is odd because the device api works as it should without any normalisation on that phone. I am using the following code

```js
var args = {
    frequency: 50,                   // ( How often the object sends the values - milliseconds )
    gravityNormalized:true,         // ( If the garvity related values to be normalized )
    orientationBase:GyroNorm.WORLD,      // ( Can be GyroNorm.GAME or GyroNorm.WORLD. gn.GAME returns orientation values with respect to the head direction of the device. gn.WORLD returns the orientation values with respect to the actual north direction of the world. )
    decimalCount:0,                 // ( How many digits after the decimal point will there be in the return values )
    logger:null,                    // ( Function to be called to log messages from gyronorm.js )
    screenAdjusted:true            // ( If set to true it will return screen adjusted values. )
}

  var gn = new GyroNorm();
    gn.init(args).then(function(){
        gn.start(function(data){
            document.querySelector('#heading').innerHTML = data.do.alpha
            document.querySelector('#beta').innerHTML = data.do.beta
            document.querySelector('#gamma').innerHTML = data.do.gamma

        });
    });
```

The same page works correctly on an android device and an iphone that I have tested it on

dorukeker, 2015-05-08:
Hello,

Which Windows Phone are you using? The phone brand and Windows version?
Also, did you try to initialise GyroNorm with the default arguments?
Such as: 

    var gn = new GyroNorm();
    gn.init().then(function(){
        gn.start(function(data){
            ...
        });
    });

I also have another remark about the code above. It is not related with the Windows Phone issue, but general. I see you have the `document.querySelector` with in the callback function of `gn.start()`.

It means it is called every 50 milliseconds. That would be heavy for the browser. Instead I would do something like this:

    // Query outside the function
    var alphaText = document.querySelector('#heading');    

    var gn = new GyroNorm();
    gn.init().then(function(){
        gn.start(function(data){
            // Use in side the callback function
            alphaText.innerHTML = data.do.alpha
            ...
        });
    });

dorukeker, 2015-08-11:
Hello,
I winder if you are having any issues?
Else I will close this issue.
Cheers,
Doruk

karneaud, 2016-03-22:
Well I have a Nokia lumina 530 running windows 8.1 using mobile IE 10 and I get nothing as well with the demo application. Though I am able get atleast the X,Y,Z values using other methods.

dorukeker, 2016-04-05:
@karneaud which other methods are you using?

karneaud, 2016-04-05:
@dorukeker I'm using the sample demo provided with the repo

dorukeker, 2016-04-06:
Ok. For my understanding:
- You are running on: Nokia Lumia 530 running Windows 8.1 using Mobile IE 10
- You run the demo file which uses Gyronorm
- You don't see any change of the alpha, beta or gamma values of gyroscope
- But you see the X,Y and Z values correctly from the accelerometer

If I got it correct:
- For the alpha, beta and gamma values, do you see `0` or nothing?
- Did you try to get the gyroscope values from the native JS events? (https://developer.mozilla.org/en-US/docs/Web/API/Detecting_device_orientation)

Cheers,
D.

karneaud, 2016-04-06:
@dorukeker I  have to admit. I had moved on from your framework and went with gyro.js But I think there was some errors concerning how you retrieved the values.

Again. I had opted to use another framework.

I note though that in Windows 8.1 you are unable to retrieve alpha beta or gamma values.

closed 2016-04-29 by dorukeker