dorukeker/gyronorm.js #10: Doesn't work with Chrome's emulation on desktop

greggman, 2015-06-25:
Chrome has device orientation emulation in their DevTools

[image]

But gyronorm fails the promise. I have no idea if there's a workaround or if this is a bug in Chrome itself that it's maybe not advertizing support? In any case it would be nice if there was a way to use gyronorm this way as it's really useful to debug.

greggman, 2015-06-25:
Hmmm, I just noticed Chrome updated to 43.0.2357.130 (64-bit) on OSX and the behavior seems to have changed. Now I have the opposite problem which is the promise resolves when I'm in standard desktop mode with no orientation available when I'd prefer it reject :disappointed:

greggman, 2015-06-25:
Ok, more ... I guess you're looking for some change in values which makes sense since the sensors usually jumble. So I found write after I start my app if I rotate the emulator within a second or so the promise resolves, if not it rejects. I guess that's fine as a workaround.

closed 2015-06-25 by greggman

dorukeker, 2015-08-11:
Hello,

Starting from v2.0.0, gyronorm is build up on FullTilt. So this is the same issue that was discussed here:
https://github.com/Full-Tilt/Full-Tilt/issues/7

I have updated gyronorm to use the latest version of FullTilt. The issue with the Chrome dev tools still remains (just like it is for FullTilt). However you can debug using [doe](https://github.com/richtr/doe).

Here you can see the gyronorm demo in doe
(https://dorukeker.github.io/gyronorm_on_doe)

Cheers,
Doruk

dorukeker, 2015-08-11:
BTW the project with flowers and butterflies is very nice :+1: