dorukeker/gyronorm.js #15: iOS Safari & Chrome - DeviceOrientation is not supported

mattheworiordan, 2016-04-02:
Hi @dorukeker 

Caveat: I am not sure this is an issue in Gyronorm, however I apologise as I am not sure where else to turn now.

I am using Chrome on Android with your library and it works fine.

However, with Chrome or Safari on iOS, all I get in the console with your demo is `DeviceOrientation is not supported`.  I thought it may be a result of a security change in iOS 9.3 so I tried serving all content over TLS / HTTPS, but that made no difference.

I get no errors or warnings, it's as if my iOS 6 device just does not have an accelerometer at all.

As you are probably looking at this sort of stuff more than most, do you have any idea what may be going on?  I've gone online and tried various `devicemotion` or `deviceorientation` demos and none of them work.

mattheworiordan, 2016-04-02:
BTW. I've also tried in an iOS Simulator, and it too doesn't work.

dorukeker, 2016-04-02:
Hi @mattheworiordan ,
I run sample code under the `/demo` folder. It ran as expected on iOS 9.3.1 on iPhone 6s.
Couple of questions/suggestions/remarks
- Are you testing with that code in the `/demo` folder? Else can you share your code?
- You mention some thing about console. I assume you get your mobile device connected to your mac developer console. And this is how you see the message? If so can you disconnect and try it. And yes you wont see any console message but you can see the numbers in the text fields changing. Indicating gyronorm is doing something.
- I have not tried with the simulator. But since it is on your desk/laptop you do not have all the physical sensors. So it might not work as expected. I suggest to test on actual devices.

Let me know if any of these help you.
And I am also curious about the project you are working on. :)
Cheers,
Doruk

mattheworiordan, 2016-04-02:
Hi @dorukeker, thanks for the quick reply, I appreciate it.

So after spending nearly 4 hours yesterday trying to figure out why the accelerometer is not working in Chome & Safari in iOS 9.3, it turns out that the problem is isolated to my phone.  I woke up this morning and tried my wife's phone and an iPad, and both worked just fine.

You can close this ticket if you want, and I'll look into the cause of the problem still.  Once I figure it out I can add it here as a reference to anyone else who has the same issue.

BTW. I just downloaded an accelerometer app on my phone, and it definitely works, so it's not a hardware problem.

PS. We'll publish our little demo soon, I'll come back and post a link as soon as I have something to show.  We're using [Ably realtime messaging](https://www.ably.io) to synchronise things in realtime using the gyro...

greggman, 2016-04-02:
I also ran it last night on my iOS 9.3 6S+ and it worked fine

dorukeker, 2016-04-03:
:) I am glad it is solved.
It'd be very nice if you can post your finding here. They will be useful for future reference.

JavaScript has an API for using the accelerometer and gyroscope data in the browser. (more details here: https://developer.mozilla.org/en-US/docs/Web/API/Detecting_device_orientation) Gyronorm works on top of these standards.

If the accelerometer app works fine but you are having problems with the browser, it might mean that the JavaScript events are not working properly. You can copy the bare code from that URL and test if the events are triggered correctly. Let me know if you need any help.

Also that Ably looks interesting. Thank you for sharing. I am currently working on an installation. I might use it there as well. I'll have a look.

Cheers,
Doruk

closed 2016-04-04 by dorukeker

mattheworiordan, 2016-04-04:
> It'd be very nice if you can post your finding here. They will be useful for future reference.

@dorukeker will do, but still at odds as to why that device cannot read accelerometer readings, whereas all my other iOS devices can.  I'm working on it and will be back with feedback for toerhs.

> If the accelerometer app works fine but you are having problems with the browser, it might mean that the JavaScript events are not working properly

Nope, all other events seem to fire just fine.

> Also that [Ably](https://www.ably.io) looks interesting. Thank you for sharing. I am currently working on an installation. I might use it there as well. I'll have a look.

Ah, great, glad to hear.  Seeing as though you've helped me out, feel free to drop me a line if you want a bump on your free quota as a thanks.

dorukeker, 2016-04-29:
@mattheworiordan How is it going? Got the demo already?
Cheers,
D.

mattheworiordan, 2016-05-09:
Sorry, no, not yet.  Waiting on the design to be applied, hopefully this week.