dorukeker/gyronorm.js #17: decimalCount 0 doesn't work

littleguga, 2016-05-16:
If I set init args decimalCount to 0. It returns 2 digit after point.

I think we need change something here: 
https://github.com/dorukeker/gyronorm.js/blob/master/lib/gyronorm.js#L108

dorukeker, 2016-05-16:
You are correct. Thanks for noticing it.
As far as I see your PR still ignores the decimalCount = 0
I suggest to change that line as follows:

```js
    if (options && options.decimalCount >= 0) _decimalCount = options.decimalCount;
```

So if you want it to return only integers you can set the decimalCount to zero.
Any comments?

littleguga, 2016-05-16:
@dorukeker which PR you watched? In #19 I changed to checking number.

I think this ```options.decimalCount >= 0``` will throw an error when no ```options``` object provided.

dorukeker, 2016-05-17:
@littleguga I merged your PR.

I will add the check to be greater than 0 (it does not break even if args is not provided). Also adding a `parseInt()` so even if the user passes a float (e.g. 2.3) library will use the integer value.

Thanks for pointing out.

closed 2016-05-17 by dorukeker

littleguga, 2016-05-17:
@dorukeker I think we need to rebuild ```/dist/```?

dorukeker, 2016-05-18:
@littleguga 
Yes :) It is built. I did not prepare a release for it. But if you check out the repo the `dist/` folder contains the new version.

Are you planning to use it on a project? Or research?

littleguga, 2016-05-18:
@dorukeker big thanks! 

I am using it my project - a little game for mobile devices in local network, when it will be done - I will publish it on github.

commit 580b5742 (2016-05-17): Merge pull request #19 from littleguga/master

fix #17
commit 953f5802 (2016-05-17): fix #17