dorukeker/gyronorm.js #20: Cannot read property 'getScreenAdjustedEuler' of null

littleguga, 2016-05-16:
I am using script from /dist/
```w.isAvailable	@	gyronorm.complete.min.js:12```
[image]

dorukeker, 2016-05-17:
Hi,
I think you are using `dist/gyronorm.min.js`. This does not include any dependencies. That is why you get an error from another library.

Please use `dist/gyronorm.complete.min.js`

Cheers,
Doruk

littleguga, 2016-05-17:
Nope, I used ```dist/gyronorm.complete.min.js```

Proof: https://jsfiddle.net/04gxf0az/

dorukeker, 2016-05-17:
Now had a closer look. I see you point. The `isAvailable()` function works only after `gn` is initialised.

```js
var gn;
var Promise = Promise || ES6Promise.Promise; //fix for promises
gn = new GyroNorm();
gn.init().then(function(){
    if (gn.isAvailable(GyroNorm.ACCELERATION)) {
        console.log("Acceleration is available");
    } else {
        console.log("Acceleration is not available");
    }
}).catch(function(e){
    console.log("Device not supported.");
    console.log(e);
})

```

There are two cases:
1) The device does not support orientation at all. In this case the promise will not resolve and you will get an error.
2) Device is said to be supporting, but the values returned from the device are not as expected. In this case `isAvailable()` function will be useful.

You can read more about it in the API here https://github.com/dorukeker/gyronorm.js/wiki/API-Documentaion#isavailable

And please note that this function is not derived from the standard, but rather from own experience and testing. The details are also in the API section.

I hope this is useful.
Cheers,
D.

closed 2016-05-17 by dorukeker