dorukeker/gyronorm.js #21: Inconsistent readings of alpha angle

Paultok, 2016-06-09:
I am using gyronorm.js on a LG G2 smartphone with recent versions of Firefox and Chrome. I am unable to get consistent readings after some rotations. For example, after the first reading, if I keep rotating the phone 90° clockwise and then back to the original position, then the alpha angle at the original position constantly decreases with a rate of ~5° per "rotation cycle".

Do you know why this happens? Is there a way to fix this? Does it depend on the phone (e.g. is it old)?

dorukeker, 2016-06-11:
Just to dobule check the issue is as follows:

Your device is lying on the table with the screen facing up. You start the Gyronorm and get the alpha value as 0. You rotate the phone around z-axis (screen still facing up). And after couple of rotations for the same position it starts to show slightly different value than 0.

Unfortunately I know the issue. In a native app the gyroscope values are super accurate. But in the web browser they are not. So they change over time. Gyronorm is build on top of the native values in the JavaScript so the results you get from Gyronorm is also shifting.

The amount of that shift differs depending on the phone model and the browser you use.

I gave some though on how to address this issue. I could not find any yet :) Any suggestions?

greggman, 2016-06-11:
> In a native app the gyroscope values are super accurate. But in the web browser they are not

If this is true on the same phone (and if this is Android/Chrome) could you please report the issue at http://crbug.com with the app that's accurate and simple web example that shows the issue?

I know the team would want to fix it

dorukeker, 2016-06-11:
Thanks @greggman I will check it. Though I am not sure if it is the browser causing the issue; or the device passing the values to the browser like this. :) It sure worths to try.

Paultok, 2016-06-11:
@dorukeker It's exactly like that. However, I cannot confirm that a native app would not suffer from the accuracy problem though, as I do not know one to try at the moment. The javascript code I am using in the browser is as simple as it could be and actually directly comes from the Gyronorm examples.

If there's anything else I can do to help to understand or fix this issue, let me know.

dorukeker, 2016-06-13:
@Paultok thanks for the info. As far as I have experienced the shift comes from the browser. Meaning even if you use low level JacvaScript API, you get this kind of behaviour. I will see if I can over come this.

@greggman btw the issue is not chrome only. The same thing happens on iOS as well. Also on the values on Safari shifts over time.

Since this is not a Gyronorm issue I am closing it. But I will go on researching if I can build something in Gyronorm to fix this.

closed 2016-06-13 by dorukeker

greggman, 2016-06-13:
My point is if you have a native app where the values don't drift and the values in Safari or Chrome drift then you should file a bug on the browser. The team browser team will be very happy for a verifiable case and be happy to fix the issue.

Paultok, 2016-06-13:
@dorukeker I can confirm that the readings from browser - without using Gyronorm - suffer from the same problem. By the way, in your experience, do you know if this issue happens to some smartphone devices or android versions only or does it always happen? What are you thinking about to solve this problem?

dorukeker, 2016-06-14:
@Paultok I have not made an extensive research on this. But I have noticed this happening regardless of the brand or OS version. Perhaps some more some less.

I will look into to see if other sensors data is available in the browser and can be used to avoid that drift.

What is your use case? Maybe we can think of a custom solution for you. Feel free to ping me on Twitter @dorukeker

Paultok, 2016-06-15:
@dorukeker my use case is very simple. I just need to read the angle on the z axis frequently (every 50ms is ok) and the read must be as precise as possible.  I need to do this in any browser, with the gyronorm.js or any other javascript code, but without installing any third-party plugin. It's really disappointing that modern browsers have this kind of problem while other native applications do not.

greggman, 2016-06-15:
Just to reiterate

Provide proof native apps don't have this problem. File a bug on the
browsers. Issue will get fixed

Paultok, 2016-06-15:
@greggman would pointing out a native app (not that I know one atm) NOT having this problem be enough to confirm that the problem exists?

Paultok, 2016-06-15:
@greggman I do not know count if this counts as a proof, but it was easy to find out one application that does not have this problem. it's called "Device Orientation" from Google Play. I made some tests and the value of the angle is the preserved as expected whenever the mobile is moved to the same position from any other position.