dorukeker/gyronorm.js #23: Callbacks fired when the event happen instead of setInterval

jalbam, 2016-07-18:
Hello,

Thank you for share your code with us!

I have been searching for a library to normalize the data returned by deviceorientation, devicemotion and MozChange events. The unique libraries I have found is this one (GyroNorm), Full-Tilt and Gyro.js.

I would like to know the reason why these three libraries use a setInterval (not even a recursive setTimeout) instead of accepting a callback function that could be called as soon as the event happens. Is it because of performance reasons? I am not sure, but maybe you took that decision because performance could be an issue since it seems you are doing many math operations and those events could fire again before your code finishes.

If there is no problem and you do not mind, I would be glad if you could add a callback function for each of those three events. By the way, I have seen you are not supporting the MozChange event. I know it is quite old but there are still some people using old browsers. The Gyro.js library supports MozChange event but does not have "game" and "world" modes.

The reason I am looking for this kind of library is because I am developing an open source framework to let other developers to create their own game engines, games, emulators and multimedia apps. And I would like to still support quite old browsers (my framework is working in IE6 which some people still use in some parts of the world although many of the features as the Device Orientation API, etc. will not be supported by IE6, of course).

If you could improve your library with those changes and you do not mind, I would be glad to use it for my project. Of course, I would list you in the credits.

To be honest, my framework already mananges those events. I just need a simple way to normalize the values returned by their event handlers between different browsers. So perhaps, instead of a library, an static object with a method or just a simple function that took the object to normalize as a parameter and returned it normalized could be better.

Thank you for all. I am sorry about the long text and my English (it is not my mother tongue).

Cheers,
 Joan

dorukeker, 2016-07-19:
Hi there.
Please see my replies below:
>>The unique libraries I have found is this one (GyroNorm), Full-Tilt and Gyro.js.

Yep. AFAIK these are the current ones. Gyro.js was the first, then Gyronorm and later FullTilt. And since FullTilt is better with the underlying maths, the 2nd version of Gyronorm I build on top of FullTilt.

> I would like to know the reason why these three libraries use a setInterval (not even a recursive setTimeout) instead of accepting a callback function that could be called as soon as the event [...]

In some devices/browsers the events can be triggered faster then the others. By using `setInterval` you make sure your callback function is called at the same frequency over different devices. So in your application you know when to expect that call.

In Gyronorm the default frequency is set to 50ms. AFAIK this is very close to the frequency of the native events. But you can reduce that time by adding `frequency:[time in ms]` to the options up on initialising the object. (https://github.com/dorukeker/gyronorm.js#options)

> If there is no problem and you do not mind, I would be glad if you could add a callback function for each of those three events. By the way, I have seen you are not supporting the MozChange event. I [...]

The current version of Gyronorm is build on top of FullTilt. And FullTilt handles the event listening. So I don't have control over what events are being listened. If I change this in the future I will consider listening for other events then.

Having said that you can still fork FullTilt and add the event your self. Be aware of the current licensing of the FullTilt though. ;) (https://github.com/adtile/Full-Tilt/issues/17)

>> I just need a simple way to normalize the values returned by their event handlers between different browsers.

The current implementation of the standards causes many issues. You can read them here https://leanpub.com/gyronormjs . I build Gyronorm to use in an installation project (https://vimeo.com/111192240) My experience then was that there is no silver bullet for all the issues. I cannot think of a simple way.

What is the aim of your frame work? Maybe there are other frame works out there? Like ThreeJS which has a VR mode as well? I am closing the issue for now but let me know if you have other questions. I will try to answer them.

closed 2016-07-19 by dorukeker

jalbam, 2016-07-31:
Thank you very much for your reply.  I will try to find a solution to normalize the events without breaking any license.