dorukeker/gyronorm.js #24: Parameter orientationBase: GyroNorm.WORLD chrome/android

firewolf, 2016-08-17:
I have problem with orientation sensor and its respect (or better - leak of respect) to GyroNorm.WORLD on android (6.0.1)/chrome (52.0.2743.98).

dorukeker, 2016-09-21:
I've just tested this, and can confirm I had the same thing on newer Android+Chrome. (iPhone behaves as expected).

Also normally if you don't define the orientationBase in the settings, it should show GAME for iPhone and WORLD for Android by default. This also does not work as expected for Android.

Gyronorm is using an underlying library (Full Tilt) to get the orientation values from the browser. The WORLD vs GAME setting is handled at that level. The same issue is mentioned here for FullTilt https://github.com/adtile/Full-Tilt/issues/22

Seems like this happens because Chrome has changed how they handle the device orientation by default. That breaks if you want world-based values. If you use game-based values both devices work like before. I will keep an eye on if/how FullTilt will address this.

For now I don't have a quick solution. I will keep this thread open for further updates.
Please add any updates on your side here; so other following can benefit from that.

Cheers,
Doruk

janrg, 2016-09-25:
I have the same problem, with Chrome on both android and iOS.

jasan-s, 2016-10-30:
could this account for the chnage? https://developer.mozilla.org/en-US/docs/Web/API/Window/ondeviceorientationabsolute

dorukeker, 2016-10-30:
I have bot tested this. My experience is that similar events are not implemented (or implemented differently) accross device+browsers. Also at the top of that page it warns it is not a standard and should not be used for production.

Having said that I have not used tried it myself. I will keep it in mind when/if I dive deeper into this issue.

jasan-s, 2016-10-30:
Im on android 5.1 using chrome 54.0.x  and I'm still getting absolute orientation alpha from the deviceOreination event. I thought this was affecting all chrome 50 and above? or is this affecting when the condition is android 6 and chrome ver > 50

dorukeker, 2016-10-31:
In the earlier versions of Chrome the default orientation base was WORLD. Meaning you get absolute values. If you want relative values, you need to set the orientation base explicitly to GAME. See the parameters you can pass when you first initialize GyroNorm object.

anjdreas, 2017-09-12:
Full Tilt project seems stagnant, there are a couple of forks that added the absolute API for google.
https://github.com/adtile/Full-Tilt/network