dorukeker/gyronorm.js #25: getting different orientation values from ios and Android

jasan-s, 2016-10-27:
i wanted to use the lib to achieve the following:  I have two geo points and want to point an arrow towards the destination geo point.  I can successfully calculate the bearing in degrees. I am currently just  using data.do.alpha but the value is different from each (ios and android) device thus the arrow points different directions as i move the phone.  What am i missing?

dorukeker, 2016-10-30:
Hello,
Since your scenario is with geolocation I assume you are using `GyroNorm.WORLD` as `orientationBase`. Google recently change how `deviceorienation` is interpreted in Chrome browser. This broke the `alpha` values in case you are using WORLD-based orientation. This was mentioned in another issue here (https://github.com/dorukeker/gyronorm.js/issues/24)

Chances are that even if you use the same iOS device and test with Safari and Chrome you will get different results.

Unfortunately it is not an easy fix that I can apply to GyroNorm (or the underlying library FullTilt). Please follow the other thread for updates on that one.

Let me know if I understood the problem correct and this addresses your problem.
Cheers,
Doruk

dorukeker, 2016-11-14:
Hello @jasan-s ,
How did it go?
I'd like to close this issue if all is clear.
Let me know.
Cheers,
Doruk

closed 2016-11-16 by dorukeker