dorukeker/gyronorm.js #27: Usage with webpack

GabrielGil, 2016-11-18:
Hello!

I am trying to use this library with webpack (ES6 import) however, something seems to be missing as the file imported is the main one specified in `package.json` which seems to not have all the needed dependencies.

Is any work on progress about this? Something I might help on?

Thank you!

dorukeker, 2016-11-19:
Hello Gabriel,
I am not very knowledgeable about the ES6 imports. And there is no current work on this.
So if you can fix and send a PR I'd be more than glad to merge. :)

Cheers,
Doruk

GabrielGil, 2016-11-23:
alright, I understand. I am currently experimenting with the library and it seems the major of the problems are the libraries.

I will be working on this definitely, so I would appreciate to keep this issue open because I am planning to send a PR at some point soon. 

Thank you!

dorukeker, 2016-11-23:
Super. I will keep it open.
Cheers!

GabrielGil, 2017-01-27:
Hi! Thank you for ping me on this. I have just tried it out on 2.0.5 and it seems that still not working right.

I have checked the PR as well, and that solutions seems not valid to me. I believe the error is in the export around [this line](https://github.com/dorukeker/gyronorm.js/blob/master/dist/gyronorm.complete.js#L1483).

The error I am getting is:

```
Uncaught TypeError: Cannot set property 'GyroNorm' of undefined
```

GabrielGil, 2017-01-27:
I made it work with AMD and CommonJS finally. 😄 

I will make a PR tomorrow after some testing. I hope it helps, and thank you again for your attention.

closed 2017-02-13 by dorukeker

botraunak, 2018-03-21:
Still facing this issue. what version should we be getting where this would be solved? i'm using `"gyronorm": "^2.0.6",`

GabrielGil, 2018-03-21:
Yes, I can confirm at version 2.0.6 the imports work just fine:

```js
// ES6+
import Gyronorm from 'gyronorm';
// Typescript
let Gyronorm = require('gyronorm');
let gs = new Gyronorm();
console.log('Gyronorm instance', gs);
```

botraunak, 2018-03-21:
@GabrielGil  Thanks for the reply. npm was fetching 2.0.4 for unknown reasons.
Related: [https://github.com/dorukeker/gyronorm.js/issues/40#issuecomment-308603450](https://github.com/dorukeker/gyronorm.js/issues/40#issuecomment-308603450)

PR #30 (2017-02-13): fix: Prevent root attachment for exports
This set of changes fixes #27 

Both AMD and CommonJS do not require the exports to be scoped globally. This is, in the window object. Actually in this kind of cases, the `root` variable is undefined on import.

Creating an export object, makes possible to import GyroNorm with tools like webpack. A factorized GyroNorm model is then ready to use. An example using ES6 with Webpack would be:

``` js
import { GyroNorm } from 'gyronorm';

let gn = new GyroNorm();
  gn.init().then(() => {
    gn.start((data) => {
      console.log(data); // GyroNorm data.
    [... cut]