dorukeker/gyronorm.js #33: Uncaught TypeError: Cannot set property 'GyroNorm' of undefined

uMaxmaxmaximus, 2017-02-14:
webpack, strict  mode;

you not correct detect root object!

need:
```javascript
var root = Function("return this")();
```

uMaxmaxmaximus, 2017-02-14:
oh, im wrong, 

https://monosnap.com/file/drOD9zSAnN3DhO7O9jO6LS5JFpqTJh.png
[image]

first anrument is not `this` is `undefined` =)))))

GabrielGil, 2017-02-22:
hi! Glad to find out more people are using Gyronorm with webpack. I have changed the exporting on febcc79d39abf1468a6feefb19b4b03ffde59732, this has been released on 2.0.6, but @dorukeker didn't build it on the complete version.

This is why I believe this does not apply to the latest version. 

Please, let me know if I can be of any help. :)

dorukeker, 2017-02-28:
Hi @GabrielGil 
I now see your PR was not included to the release.
I remade the release. It should be solved now.

Can you have a look and let me know if all works fine?

Cheers,
Doruk

closed 2017-02-28 by dorukeker

GabrielGil, 2017-02-28:
Yes, it is perfect now. I had actually mentioned in my pr (https://github.com/dorukeker/gyronorm.js/pull/30#issuecomment-275853161) that just needed the build and version number updates.

Thank you one more time @dorukeker