dorukeker/gyronorm.js #36: DeviceOrientation && DeviceMotion is not supported

stephanedemotte, 2017-03-28:
Hi !

Thanks for this useful library.

I try to use it in a WebView in React Native.

In IOS gyronorm return `DeviceOrientation is not supported`
In Android `DeviceMotion is not supported`

But `DeviceMotionEvent` and `DeviceOrientationEvent` exist on both.

Any idea ?

Thank you !

dorukeker, 2017-04-02:
Hi there,
Gyronorm is using the JS API from the browser; and it is meant to be used with in the web browser; not native apps.

I don't have much experience with React. But since you are packing to make native apps; I assume there are better (more accurate) ways to access devicemotion and deviceorientation data?

A quick Google search give me this: https://github.com/paramaggarwal/react-native-device-motion

Let me know if this was helpful.
Cheers,
Doruk

dorukeker, 2017-09-10:
Hello @stephanedemotte I am going to close this issue, assuming you have found another way via React native. Please feel free to reopen it if you have more questions about Gyronorm

closed 2017-09-10 by dorukeker

stephanedemotte, 2017-09-10:
yes thanks for the support !