dorukeker/gyronorm.js #38: Unable to get gyronorm working in Firefox Android

aniket-21, 2017-05-10:
Hi, I am unable to get this library working in Firefox Android v53 and Firefox beta 54.
But from other issues I see people have been using this on Android Firefox as well.

Do you know what might be the reason?

dorukeker, 2017-05-18:
Hi there. It can be a device thing. Can you check if the native Wenb API DeviceOrientation is working as expected on that device? It is descripbed here: https://developer.mozilla.org/en-US/docs/Web/API/Detecting_device_orientation

If it is not supported by the device you can handle it in the `catch` section of the promise.
This does not solve the issue but at least you can fallback gracefully ;)

Let me know how it goes.

aniket-21, 2017-05-18:
yes native API works. Rather gyronorm works in Chrome but not in firefox

dorukeker, 2017-06-05:
Hello,

I don't have an Android phone (or a Motorola in this case).
I checked the Firefox on iOS and the demo run as expected.

Anyone else can test on Android Firefox?

Cheers,
Doruk

dorukeker, 2017-09-10:
Hello,
Tested the Gyronorm demo (https://dorukeker.github.io/gyronorm_samples/demo) on Samsung Galaxy S6 with Android 5.0.2 and it is working as expected.

For now I am closing this issue. Feel free to re-open it if you have questions about Gyronorm.

closed 2017-09-10 by dorukeker