dorukeker/gyronorm.js #40: Installing v2.0.6 via NPM installs v2.0.4

glennflanagan, 2017-06-01:
I've just tried installing this library via NPM and both `npm install gyronorm -S` and `npm install gyronorm@2.0.6 -S` are installing the library, and adding v2.0.6 as the dependency in package.json file, but the code downloaded and in the node_modules folder is the code for v2.0.4.

I've managed a workaround by adding this as the dependency in my package.json
```
"gyronorm": "git://github.com/dorukeker/gyronorm.js.git#v2.0.6",
```

Can someone else try installing v2.0.6 via NPM and check the node_module folder to confirm this is an issue?

Thanks

dorukeker, 2017-06-02:
Hi there,

I had the same issue before and it was fixed. (https://github.com/dorukeker/gyronorm.js/issues/34) Apparently for some reason it is back.
I will check with NPM and let you know.

Cheers,
Doruk

dorukeker, 2017-06-05:
Hello,
On Mac it installs the correct version (2.0.6). On PC it still installs older one (2.0.4)
It tells me this can be computer specific.

Any ideas? Did it et fixed for you?

Cheers,
Doruk

glennflanagan, 2017-06-05:
Hey @dorukeker,

I was using a Mac when I was running into this issue. Running on Node v6 NPM v3. 

I've not checked on another machine yet to see if it's consistent. I will try to do that soon for you to see if I can repeat the issue.

glomotion, 2017-06-15:
Hi there! 
Also having this issue. 
OSX running node@6.11.0 and yarn version 0.24.6

For now i've reverted to using the github repo url instead.

dorukeker, 2017-06-15:
Hi There,

I had a long email traffic with NPM about this. We tried to clear the npm cache etc.
But could not figure out the actual reason. In some computers (both mac and pc) it still installs an older version.

For now I can think of two options:
1) using the github url (as suggested by @glomotion )
2) using bower - `$ bower install gyronorm`

Hope this helps.
Cheers,
Doruk

glomotion, 2017-06-15:
Nps - should be fine for now.

BTW, unrelated - but *thank you* from the bottom of my heart, for making this available on npm and in MIT. 

Having to use bower just for fulltilt was killing me. :P

dorukeker, 2017-06-16:
@glomotion I am glad you find it useful. :)

dorukeker, 2017-09-10:
Hello,
Just a feedback:
I still have this issue PC only. Super weird and the NPM people could not find out what is going on. I assume it will be fixed with the next version push. For now please use the workaround mentioned by @glennflanagan 

Cheers,
Doruk