dorukeker/gyronorm.js #43: require('GyroNorm') returns Object, not constructor

broofa, 2017-07-25:
When building with WebPack, `require('GyroNorm')` returns an Object, not a Function:

```
> console.log(require('GyroNorm'))

Object {GyroNorm: function}
```

Thus, trying to do `new GyroNorm()` per the README ends up throwing an error.

broofa, 2017-07-25:
FWIW, workaround is to do `const GyroNorm = require('GyroNorm').GyroNorm;`

dorukeker, 2017-09-10:
Hi there!
Glad that you figured out a work around.
I am not very knowledgable about webpack, hence the error. With the next push I will take this into considerations.

I keep the issue open so others can also know about this.

Cheers,
D.

botraunak, 2018-03-21:
@dorukeker @broofa It's failing for me with webpack in strict mode.
[image]
[image]

Related #27

GabrielGil, 2018-03-21:
Webpack works always in "strict mode", better said it enables this mode on each module. 

The problem in your case seems to be the version you're using. Based on your screenshot, the version you're using is 2.0.4. Now the loader looks like this:

```js
/**
 * JavaScript project for accessing and normalizing the accelerometer and gyroscope data on mobile 
   devices
 *
 * @author Doruk Eker <doruk@dorukeker.com>
 * @copyright Doruk Eker <http://dorukeker.com>
 * @version 2.0.6
 * @license MIT License | http://opensource.org/licenses/MIT
 */
var e = {
  GyroNorm: factory(),
};
if (typeof define === 'function' && define.amd) {
  define(function() {
    return e;
  });
} else if (typeof module === 'object' && module.exports) {
  module.exports = e;
} else {
  root.GyroNorm = e.GyroNorm;
}
```

botraunak, 2018-03-21:
@GabrielGil Thanks! Solved it, I'll leave the link to the npm issue here if anyone else runs into this.
Related: [https://github.com/dorukeker/gyronorm.js/issues/40#issuecomment-308603450](https://github.com/dorukeker/gyronorm.js/issues/40#issuecomment-308603450)