dorukeker/gyronorm.js #44: no consistent values for accelerometer (iphone7 vs nexus5)

fabianburghardt, 2017-08-10:
Hi! I used gyronorm.js to get consistent values for the acceleration accross different devices. The gyronorm function puts out different values than the standard system values, but there's still a big difference between, for example, iPhone7 and Nexus 5 acceleration values.

Here's a fiddle with the code:
https://jsfiddle.net/mo2owLf6/2/

Here's the same code as a demo to check on smartphones: 
http://demo.sansho.studio/acc

Is there any chance to get the values consistent?
Thanks!

dorukeker, 2017-08-10:
Hi there,
I dont have access to any of the devices. I test the link on iPhone 6S using Safari.
The values are very similar (72.9 vs 73.1). I think this is acceptable.

Below are some questions/suggestions. But before that I have to say it is a mess how different devices and browsers handle these values. Gyronorm addresses some of these differences; unfortunately not all. You can read about it here https://leanpub.com/gyronormjs

Those cases will need additional assumptions depending on the logic of individual app. This you need to implement on your side.

Here are some questions/suggestion:
Does any of the phones give you correct values? Like I am having with 6S.
Did you try using the acceleration including gravity instead dm.gy instead of dm.y
Can you check if the device is returning correct values outside of browser. With an application like this https://itunes.apple.com/us/app/sensor-kinetics/id579040333?mt=8 

I'd suggest to test it with more devices and see what is the consistent value over as many devices as possible. Then you can pinpoint the reason for different values.

Let me know if any of these help.
Cheers,
Doruk

fabianburghardt, 2017-08-10:
Hi Doruk,
thanks for the quick reply!

I just started working on a list with both values on different devices. Quite interesting!

https://docs.google.com/document/d/1PeSqsbWIpaGbV10OUfobyi-TMK2PDvtcQTNX3gHJGSg/edit?usp=sharing

dorukeker, 2017-08-11:
That is a great list indeed!

I see indeed the devices return different values. I think you can address this in the application logic. Not sure what your use case is; so no concrete suggestions.

Since it is not directly related with Gyronorm I will close those issue.
Feel free to go on posting here and/or re-open it.

And I am very interested about the use case and how/if you find a solution.

closed 2017-08-11 by dorukeker