dorukeker/gyronorm.js #45: acceleration normalized to direction of motion

arjunroychowdhury, 2017-09-03:
Hi, thanks for this excellent library. My use case is to detect acceleration values relative to the direction of motion in a car. In other words, the device could be kept in any orientation but I'd like the x/y/z to be normalized in the direction of motion. What might be the best settings to use ? So far I am using the following options (which is not producing intended results - depending on how my phone is oriented acc x/y/z report different values when I move)

```
let gyroNormArgs = {
    logger: function (data) {console.log ("Gyr-log:"+JSON.stringify(data))},
    gravityNormalized:true, 
    orientationBase:GyroNorm.WORLD,
    frequency:1000,
    screenAdjusted:true,
  }
```

dorukeker, 2017-09-10:
Hello there,
I am glad to see the library being used in different use cases.
Couple of remarks to clarify the use of parameters and the return values:

The parameters you pass are correct.

orientationBase only efects the deviceOrientation values; alpha, beta and gamma values. The acceleration values stay with respect to the axis' of the device. These values are returned like this from the JS API and GyroNorm passes them to your application.

What you are trying can be achieved by some calculation. Something like... getting the orientation world based and accelerationIncludingGravity and using it to calculate the acceleration in world axis. Gyronorm does not contain such calculation at the moment. So you need to implement on your side. You can also implement within Gyronorm and send a pull-request ;) I'll be more than glad to merge.

One last thing: you need to use the accelerationIncludingGravity values.

Let me know if anything is not clear.
Cheers,
Doruk

arjunroychowdhury, 2017-09-10:
Thanks @dorukeker 

I did a bit of reading. It looks like if I normalize the values of `gx` `gy` and `gz` (including gravity) with the rotation matrix, I'll get earth co-ordinates.

Can you check if this look ok to you and if I need to do any other conversions with the data your library provides?

```
rotationMatrix = getRotationMatrix(data.dm.alpha, data.dm.beta, data.dm.gamma)
accEarthX = rotationMatrix[0] * data.dm.gx + rotationMatrix[1] * data.dm.gy + rotationMatrix[2] * data.dm.gz
accEarthY = rotationMatrix[3] * data.dm.gx + rotationMatrix[4] * data.dm.gy + rotationMatrix[5] * data.dm.gz
accEarthZ = rotationMatrix[6] * data.dm.gx + rotationMatrix[7] * data.dm.gy + rotationMatrix[8] * data.dm.gz
```

And `getRotationMatrix` is:

```
// credit:https://stackoverflow.com/a/36662093/1361529
getRotationMatrix(alpha, beta, gamma) {
  _z = alpha;
  _x = beta;
  _y = gamma;

  cX = Math.cos( _x );
  cY = Math.cos( _y );
  cZ = Math.cos( _z );
  sX = Math.sin( _x );
  sY = Math.sin( _y );
  sZ = Math.sin( _z );

  out[0] = cZ * cY + sZ * sX * sY,    // row 1, col 1
  out[1] = cX * sZ,                   // row 2, col 1
  out[2] = - cZ * sY + sZ * sX * cY , // row 3, col 1

  out[3] = - cY * sZ + cZ * sX * sY,  // row 1, col 2
  out[4] = cZ * cX,                   // row 2, col 2
  out[5] = sZ * sY + cZ * cY * sX,    // row 3, col 2

  out[6] = cX * sY,                   // row 1, col 3
  out[7] = - sX,                      // row 2, col 3
  out[8] = cX * cY                    // row 3, col 3

return out
};
```

arjunroychowdhury, 2017-09-10:
Okay, so I ran this code, and it looks like I am not doing it right. When I hold the phone perpendicular to  the floor, I see gravity on Y. If I lay it flat, I see gravity on Z.

Here is an example of the output when holding the phone perpendicular to my table

```
2017-09-10 08:29:15.055417-0400 TripData[6027:2061953] *** EARTH DATA X=2.2943273132349287, Y=9.534772910579633 Z=-0.1909652413948768
2017-09-10 08:29:16.056623-0400 TripData[6027:2061953] *** EARTH DATA X=-4.630849432456284, Y=8.649729578711812 Z=-0.24064860082132544
2017-09-10 08:29:17.056643-0400 TripData[6027:2061953] *** EARTH DATA X=-1.9508303521644637, Y=9.600077721852953 Z=-0.305562876437087
2017-09-10 08:29:18.071261-0400 TripData[6027:2061953] *** EARTH DATA X=-0.017282406466729114, Y=9.58242138261782 Z=-2.0579362877347576
2017-09-10 08:29:19.073391-0400 TripData[6027:2061953] *** EARTH DATA X=-0.7562998091905141, Y=9.67232563579337 Z=-1.4375769175222382
2017-09-10 08:29:20.088494-0400 TripData[6027:2061953] *** EARTH DATA X=3.014021667418713, Y=9.29277056085625 Z=-0.8659033962365313
```

Here is an example of holding the phone flat on my table, facing up
```
2017-09-10 08:30:29.273031-0400 TripData[6030:2062680] *** EARTH DATA X=0.3213661367499145, Y=-0.30880129933902706 Z=-9.710152705476727
2017-09-10 08:30:30.282963-0400 TripData[6030:2062680] *** EARTH DATA X=-0.8330555159493345, Y=-0.35403901361236245 Z=-9.678061525129234
2017-09-10 08:30:31.288030-0400 TripData[6030:2062680] *** EARTH DATA X=-0.25170701382938837, Y=-0.3217699399585805 Z=-9.711684080782703
2017-09-10 08:30:32.293732-0400 TripData[6030:2062680] *** EARTH DATA X=-0.30199915826823015, Y=0.9298620373824118 Z=-9.681056404124517
2017-09-10 08:30:33.294888-0400 TripData[6030:2062680] *** EARTH DATA X=0.31606866523909627, Y=0.3622808441701922 Z=-9.70845266707323
2017-09-10 08:30:34.301882-0400 TripData[6030:2062680] *** EARTH DATA X=-0.24551748429710923, Y=0.9452959329108171 Z=-9.671108352413734
2017-09-10 08:30:35.284257-0400 TripData[6030:2062680] *** EARTH DATA X=0.9266874306677086, Y=-0.854412045314473 Z=-9.648364134021065
2017-09-10 08:30:36.301415-0400 TripData[6030:2062680] *** EARTH DATA X=0.8516336100663204, Y=0.4447390757116382 Z=-9.682826413229794
```

dorukeker, 2017-10-11:
Hello there @arjunroychowdhury 
Wow that is some maths. Frankly speaking I am not very good at the matrix operations.
Having said that I see in the code you are using the angles from the deviceMotion

```
let alpha = gdata.dm.alpha;
let beta = gdata.dm.beta;
let gamma = gdata.dm.gamma;
```

Which are the change rates in these angles; not actual orientation. The orientation are in the deviceOrientation. Like this:

```
let alpha = gdata.do.alpha;
let beta = gdata.do.beta;
let gamma = gdata.do.gamma;

// do instead of dm

```

Can this be what you are looking for? Also try to switch orientation base as you mentioned. (WORLD or GAME) Not sure which one should give you the result you are looking for.

Cheers,
Doruk

ziqbal, 2017-10-26:
Hi All,

Did you figure this out?

I'll be attempting this as well and would like to save some some time and troubles.

marcusx2, 2021-06-23:
@arjunroychowdhury @ziqbal Anyone figure this out?

ziqbal, 2021-06-24:
@marcusx2 I never found the time to check the last feedback from dorukeker. Maybe it works.
My advice is that if you're targeting a single device then you should use whatever is provided on that platform, otherwise a generic library to do absolute orientation on all devices and platforms is going to be a serious PITA. Especially when libraries become out of date and "features" become deprecated and obsolete.