dorukeker/gyronorm.js #47: shake to undo gets triggered

vin-ni, 2017-10-11:
I'm filling out an input field, before starting a game that is played by shaking the phone. Unfortunately, on iOS, this triggers the « Shake to Undo».

How can I add a event.stopPropagation(); to the original event, to stop this behaviour?

Thanks

dorukeker, 2017-10-11:
AFAIK Shake-to-undo is OS level. I dont think you can control it from the browser.
Why don't you have two HTML pages? :)

Get the input in the first. Enter/submit gets you to the second page where you have the game.

Question is not directly related to the library so I am closing it as an issue. Feel free to continue the conversation though.

Cheers,
D.

closed 2017-10-11 by dorukeker

vin-ni, 2017-10-12:
hmm, unfortunately I need to stay on the same page.
I found some sources where they said that I could prohibit this with stopPropagation somehow.
I'll do some tests.

dorukeker, 2017-10-13:
A quick search:
https://stackoverflow.com/questions/6520125/how-to-disable-ios-shake-to-undo-in-a-webapp The second reply here can be use full.

Cheers,
D.

vin-ni, 2017-10-13:
Yes, I finally used the third answer, which worked perfectly (I dont need emoji)
https://stackoverflow.com/a/43481473/1779355

dorukeker, 2017-10-13:
Glad that it worked!