dorukeker/gyronorm.js #48: Rotation rate inconsistent between Safari on iOS 11 and Chrome on Android 7

zachrattner, 2017-10-17:
I'm running a sample app based off the readme example and logging `data.dm.alpha`, `data.dm.beta`, and `data.dm.gamma`. To do a simple smoke test, I'm stacking my iPhone 6S (running iOS 11.1 and Safari) on top of my Galaxy S8 (running Android 7.0 and Chrome) and pick the two phones up with one hand off the desk and then place them back down. I see significantly more extreme values on Safari than Chrome.

On Android, I see the following data logged: 

```
1508279962612: [-0.26, -0.1, -0.06]      
1508279962912: [0.01, 0.11, 0.03]      
1508279963210: [0.18, 0.29, 0.02]      
1508279963510: [1.5, -0.31, -0.31]      
1508279963810: [3.35, -0.38, -0.56]      
1508279964111: [2.07, -0.27, -0.13]      
1508279964408: [0.04, -0.46, -0.09]      
1508279964710: [-0.57, 0.84, 0.11]      
1508279965011: [-1.72, 0.63, 0.52]      
1508279965312: [-3.31, 2.46, 0.67]      
1508279965610: [-1.21, 0.38, 0.3]      
1508279965909: [0.22, 0.23, -0.07]      
1508279966209: [-0.1, -0.11, -0.01]      
1508279966509: [0, 0.01, 0.02]      
1508279966809: [0, 0.03, 0.01]      
1508279967109: [-0.04, 0, 0.02]      
1508279967409: [0.32, 0.25, 0.02]      
1508279967662: [0.4, 0.9, -0.24]      
1508279967960: [-0.95, -0.38, 0.4]      
1508279968261: [-0.06, 1.47, -0.33]      
```

On Safari, I see more extreme values: 

```
1508279962535: [-4.84, -0.74, -2.26]
1508279962790: [-6.09, 6.33, -6.77]
1508279963043: [-0.09, 1.68, 2.17]
1508279963295: [-1.8, -1.31, 1.18]
1508279963548: [0.34, 1.13, 1.56]
1508279963803: [-0.45, 2.03, 0.77]
1508279964059: [-8.48, -4.61, 0.22]
1508279964311: [-0.4, 0.22, 2.41]
1508279964564: [11.99, 12.22, 2.87]
1508279964818: [44.61, -21.23, -17.07]
1508279965074: [93.49, -27.45, -25.17]
1508279965327: [186.64, -25.12, -30.42]
1508279965578: [116.26, -16.13, -19.9]
1508279965831: [52.35, -27.42, -14.54]
1508279966086: [-17.65, 2.11, 1.24]
1508279966342: [-73.73, 18.36, 18.02]
1508279966594: [-83.56, 36.91, 30.38]
1508279966847: [-202.29, 147.12, 52.14]
1508279967102: [-98.39, 31.99, 20.51]
1508279967357: [-26.92, -0.11, -6.16]
1508279967610: [1.29, -8.9, 3.31]
1508279967864: [-3.2, -4.63, -1.47]
1508279968118: [0.66, -0.36, -0.4]
1508279968374: [1.27, -0.55, -1.13]
```

Sample app code (including gyronorm from `https://dorukeker.github.io/gyronorm_samples/demo/gyronorm.complete.min.js`):

```
let gn = new GyroNorm();

gn.init().then(function(){
    gn.start(function(data) {
                        console.log(new Date().getTime() + ": [" + data.dm.alpha + ", " + data.dm.beta + ", " + data.dm.gamma + "]");
    });
});
```

dorukeker, 2017-10-20:
Hi there,
I have not had the chance to test on the actual devices. I will when I have the time.
Having said that, here are some questions/remarks:

- Which browser are you using on Android?
- `data.dm.xxx` gives you the change in the rotation; not the rotation itself. I assume this is what you wanna get correct? If you want just the rotation you should use `data.do.xxx`
- I see on Safari some `alpha` value is around 200. That means that the device was rotated in a short amount of time almost 3/4 around itself. Was that really the case?
- Finally can you try this with vanilla values from JS API?

Curious how it goes.
Cheers,
Doruk

zachrattner, 2017-10-25:
@dorukeker Thanks for the reply. Responses below:

- Chrome 61.0.3163.98 on Android 7.0.0
- Correct, the use case I'm looking for is more concerned with the rate of change than the actual rotation itself
- The device was picked up from lying face-up flat on a desk to facing me (similar to what you would do if you got a text and checked the screen). I repeated this motion three times. 
- Do you have a code snippet that uses the vanilla values?

Thanks,
Zach

skrichten, 2017-11-20:
I have just observed the same behavior with an older version of Android, and yes I get the same from vanilla values.  The values on Android/chrome seem more correct/logical to me.  The values on ios seem almost like they are off by an order of magnitude.