dorukeker/gyronorm.js #52: Gyronorm within Web Worker

StephenCathcart, 2018-05-16:
Hi,

I'm in need of running gyronorm.js within a web worker. I'm running gyronorm within a cordova application and the main background thread slows down, updating once per second. If I place gyronorm within a web worker I'm getting an error thrown saying that the window object us undefined (due to web workers having no access to the window object). Is there a way around this to get the script to work in that context? Cheers

dorukeker, 2018-05-16:
Hello there. Since you mention Cordova I assume you are packing the app to run as a native app on the mobile devices; is that correct?

Gyronorm is intended only for web applications running in the browser. I assume there are better (and more accurate) solutions to access the deviceMotion and deviceOrientation data in a Cordova app.

Quick Googloing return there:
https://www.npmjs.com/package/cordova-plugin-gyroscope
https://cordova.apache.org/docs/en/latest/reference/cordova-plugin-device-motion/

dorukeker, 2018-05-16:
Perfect! I close the issue for now.
You can still update for future reference.
Good luck!

closed 2018-05-16 by dorukeker

StephenCathcart, 2018-05-17:
As suspected the issue was down to a combination of chrome throttling timers and also iOS exiting the application when suspended. To get this library working nicely in an Ionic setting then add this script to the index.html file:

`<script src="assets/js/hack-timer.min.js"></script>`

Next, update the config.xml file as follows:

```
<platform name="ios">
// other stuff
    <config-file parent="UIApplicationExitsOnSuspend" target="*-Info.plist">
        <true />
    </config-file>
</platform>
```
Using the value of true for that property name seemed a bit counter-intuitive to keep the application running but that works. After these changes I was able to get back to capturing motion and orientation data at 50hz even with the phone locked.

Thanks Doruk Eker.

(P.S the reason for not using the cordova gyroscope and motion plugins is due to the official docs deprecation notice stating with the W3C Device Motion and Orientation API now being supported on iOS, Android and Windows devices, the plugins are no longer needed)

https://www.npmjs.com/package/cordova-plugin-device-motion