dorukeker/gyronorm.js #54: Acceleration is not consistent between iOS and Android

zcfan, 2018-06-14:
I found iOS and Android acceleration value has different sign, and it seems gravityNormalized option only correct accelerationWithGravity(gm.gx/gm.gy/gm.gz) but not acceleration(gm.x/gm.y/gm.z).

I made a plot for this situation to clarify my idea:

[image]

Those two green box shows that the `gm.y` and `gm.z` in iOS and Android are negative to each other.

Change line 350 - 355 to following could give consistent value, but I'm not sure I understand this library completely. I would like to make this a PR if you this's a correct fix.

    // Normalize gravity
    if (_gravityNormalized) {
      snapShot.dm.gx *= _gravityCoefficient;
      snapShot.dm.gy *= _gravityCoefficient;
      snapShot.dm.gz *= _gravityCoefficient;
    }
    // Change above to:
    if (_gravityNormalized) {
      snapShot.dm.gx *= _gravityCoefficient;
      snapShot.dm.gy *= _gravityCoefficient;
      snapShot.dm.gz *= _gravityCoefficient;
      snapShot.dm.x *= _gravityCoefficient;
      snapShot.dm.y *= _gravityCoefficient;
      snapShot.dm.z *= _gravityCoefficient;
    }

dorukeker, 2018-06-14:
Yep! you are correct please make the PR and I will merge.
Thanks!
D.

zcfan, 2018-06-15:
@dorukeker Actually I found that gravityNormalized option can't be set by user at all, it's always true. Will make a PR later 😂

dorukeker, 2018-06-16:
Thanks! I merged the PRs

closed 2018-06-16 by dorukeker