dorukeker/gyronorm.js #57: Callback not behaving properly in Chrome

nicu-chiciuc, 2018-06-19:
In Chrome the function is called with all the values set to 0 instead of calling the catch function.

dorukeker, 2018-06-22:
Can you give more info about the version and OS.
Also please try this basic demo http://dorukeker.github.io/gyronorm_samples/demo/ on the same browser.

nicu-chiciuc, 2018-06-25:
The values are set to 0 in:
Windows, Chrome 67.0.3396.87

The values are not set in:
MacOS, Chrome 67.0.3396.87
Windows, Firefox 60.0.2
MacOS, Firefox 60.0.2

dorukeker, 2018-06-25:
I assume you are on desktop? Is that correct?

Gyronorm uses native java script APIs of deviceorientation and devicemotion. Normally these are meant for mobile devices.

For the desktop (laptop): If the laptop has gyroscope and accelerometer sensors and if the browser has access to them via JavaScript then perhaps you can get accurate values via gyronorm. But this is not the target usecase.

If your use case is mobile devices; the. I suggest to test on target devices. If you target laptops/desktops I would not suggest to count on device orientation and motion via JavaScript.

Please let me if this is sufficient info for you. Then I will close the issue.

nicu-chiciuc, 2018-06-25:
I was working on a project which required that the desktop use the mouse and the mobile use the gyroscope. The problem was that I relied on the `catch` as explained in the documentation:
```
}).catch(function(e){
  // Catch if the DeviceOrientation or DeviceMotion is not supported by the browser or device
});
```
It would be possible of course to use something else to check if the device is a mobile or not but it seems like this is something that should be addressed since it behaves unexpectedly.
I could look more into this.

dorukeker, 2018-06-25:
I’d use this one https://github.com/dorukeker/gyronorm.js/wiki/API-Documentaion#isavailable to check if it is available. Then use the start function.