dorukeker/gyronorm.js #58: Support for absolute orientation

AlexZeller, 2018-06-26:
Hello,
I am using gyronorm.js to try to build a VR/AR Application in combination with Cesium.js (A 3D mapping library). The idea is to discover the terrain around you by pointing your phone. 

The problem is, that I need the absolute orientation values with respect to the Earth’s coordinate frame.  I have tried GyroNorm.WORLD  as well as GyroNorm.GAME but both return relative parameters.

Is there support for absolute orientation or is it a limitation of the browser?

Thanks

dorukeker, 2018-06-26:
Not sure which browser you are using; so it might be due to the browser.
Also please have a look in the API docs how the GyroNorm.WORLD and GyroNorm.GAME works.

Having said those: if you wanna have AR/VR on the browser I recommend that you take a look at webvr or A-frame. These are much more up-to-date compared to gyronorm. Web AR is also available but not widely supported main stream browsers yet.

AlexZeller, 2018-06-27:
Sorry, I forgot to mention. I now tested on Chrome and Firefox on both an IOS and Android Device. 

I implemented the orientation base as follows:

```
var args = {			
    orientationBase:GyroNorm.WORLD		
};
gn.init(args).then(function(){
    gn.start(function(data){
    }
}
```
For some reason I do get absolute orientation on all browsers values now. So it was some mistake on my part (maybe an old version of my javascript was cached).

Sorry for bothering you. Thanks for your help.

closed 2018-06-27 by AlexZeller

dorukeker, 2018-06-27:
You are welcome! @AlexZeller